vial-kb/vial-qmk #4: [Feature Request] Allow vial config to be done at a keyboard level.

Pieterv24, 2021-04-18:
# Proposal

Currently to support vial with a keyboard the following needs to be done
- Add VIAL_KEYBOARD_UID to the config.h in the keymap folder
- Add vial.json to the keymap folder

What would be nice to have is to be able to add a keyboard variant that contains both the vial.json and config.h.

For example currently when you've got 2 keymaps supporting VIAL it look like this.
This would require the vial.json and config.h to be created 2 times

<details><summary>Folder structure now</summary>
<pre>
keyboards/
├─ pieterv24/
│  ├─ testboard/
│  │  ├─ keymaps/
│  │  │  ├─ default/
│  │  │  │  ├─ keymap.c
│  │  │  ├─ pieterv24/
│  │  │  │  ├─ config.h
│  │  │  │  ├─ keymap.c
│  │  │  │  ├─ rules.mk
│  │  │  │  ├─ vial.json
│  │  │  ├─ vial/
│  │  │  │  ├─ keymap.c
│  │  │  │  ├─ rules.mk
│  │  │  │  ├─ vial.json
│  │  │  │  ├─ config.h
│  │  ├─ config.h
│  │  ├─ info.json
│  │  ├─ readme.md
│  │  ├─ rules.mk
│  │  ├─ testboard.c
│  │  ├─ testboard.h
</pre>
</details>

My proposal would be to allow the following

<details><summary>Folder structure proposal</summary>
<pre>
keyboards/
├─ pieterv24/
│  ├─ testboard/
│  │  ├─ keymaps/
│  │  │  ├─ default/
│  │  │  │  ├─ keymap.c
│  │  │  ├─ pieterv24/
│  │  │  │  ├─ keymap.c
│  │  │  │  ├─ rules.mk
│  │  ├─ vial/
│  │  │  ├─ vial.json
│  │  │  ├─ config.h
│  │  ├─ readme.md
│  │  ├─ rules.mk
│  │  ├─ testboard.c
│  │  ├─ testboard.h
</pre>
</details>

In this configuration the idea would be that you could compile each keymap as a vial compatible keymap.
For example like this:
`make pieterv24/testboard/vial:default`
`make pieterv24/testboard/vial:pieterv24`

or without vial like:
`make pieterv24/testboard:default`
`make pieterv24/testboard:pieterv24`

## Feature Request Type

- [x] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [x] New behavior

xyzz, 2021-04-18:
> Add VIAL_KEYBOARD_UID to the config.h in the keymap folder

As per docs, you should add these to the keyboard config.h: https://get.vial.today/gettingStarted/porting-to-vial.html#4-set-up-and-add-unique-keyboard-id. Then, it will be shared across all keymaps. So I'm not sure I understand the problem here.

Pieterv24, 2021-04-19:
it was mostly aimed at allowing the vial.json to be put on a keyboard level aswell.
Not sure if that breaks anything how you invisioned it

Pieterv24, 2021-04-19:
What I pimarily envision is that you can add a extra keyboard definition inside the keyboard, which contains all the code for vial.
so you could have standard keymaps without any VIAL relations.
An a keyboard revision that contains all VIAL related settings.

And then being able to compile a standard keymap agains the vial revision of the board to create a vial compatible firmware variant.

xyzz, 2021-07-21:
Since this is a feature suggestion and not a bug/another issue in Vial, I've gone ahead and moved it to the dedicated feedback website at https://feedback.vial.today/suggestions/198784/allow-vial-config-to-be-done-at-a-keyboard-level.

closed 2021-07-21 by xyzz