vial-kb/vial-qmk #94: [Bug] No OLED and no Encoder on Sofle keyboard

ricrug, 2021-12-17:
After flashing the keyboard with the firmware vial, the displays stay off and the encoders do not work when switching on.

 - Keyboard:  Sofle
 - Operating system: Linux
 - AVR GCC version: 10.2.0
 - ARM GCC version: 10.2.0

xyzz, 2021-12-17:
@kirbysayshi could you take a look?

kirbysayshi, 2021-12-17:
I’ve used this for both sofle v1 and v2, and the OLEDs have worked for both. Could it be a seating or connection issue? Do the OLEDs work when using QMK or via?

ricrug, 2021-12-18:
> I’ve used this for both sofle v1 and v2, and the OLEDs have worked for both. Could it be a seating or connection issue? Do the OLEDs work when using QMK or via?

yes, it works correctly with via and the default firmware.
Mine is a sofle v2. We tried it in two, on two different keyboards and from two different PCs.

kirbysayshi, 2021-12-18:
My only guess is that vial changed the OLED interface and now the hook is not being called: https://github.com/vial-kb/vial-qmk/commit/9d235d4fc5c34f372b0c393f41686b5ef4f8dc13#diff-183c0f65ab3d6e9b42d26f0ce05abd45affc4d1a415d6753cdcf09d3271e6a9dR41

if you change oled.h in Sofle to return bool I bet it will work

ricrug, 2021-12-19:
I think... it is not the only change... there's somthing wrong in the define (I forced it manually into oled.c) and it works... 
I'm still not able to change the logo... even if I changed the qmk_logo array.

xyzz, 2022-03-30:
Should be fixed by #136

closed 2022-03-30 by xyzz