vial-kb/vial-qmk #117: 编码器不能使用 Encoder cannot be used

jeromeli002, 2022-02-09:
多个不明确 映射到矩阵位置的，相同的配置 qmk下编译出来编码器可以用，也能via修改，vial-qmk直接编码器无反应不行，已经添加了VIAL_ENCODERS_ENABLE = yes

Multiple encoders that are not explicitly mapped to the matrix position can be used or modified via the encoder compiled under the same configuration qmk. The via qmk direct encoder has no response, and via has been added_ ENCODERS_ ENABLE = yes

xyzz, 2022-02-09:
Please add a link to your keyboard and `vial` keymap source code.

jeromeli002, 2022-02-10:
> 请添加指向您的键盘和`vial`键盘映射源代码的链接。
4个编码器 在qmk里面编译出来是完全正确可以使用的，但是via-qmk编译出来编码器不能使用，右键添加via_ENCODERS_ENABLE=yes进去了

The 4 encoders compiled in qmk are completely correct and usable, but the encoder compiled by via-qmk cannot be used, right click and add via_ENCODERS_ENABLE=yes into it

bool encoder_update_user(uint8_t index, bool clockwise) {
		uint8_t  layer_now                 = biton32(layer_state);
		uint16_t encoder_cw_keycode        = keymap_key_to_keycode(layer_now,(keypos_t){.row = 2,.col = 5});
		uint16_t encoder_ccw_keycode       = keymap_key_to_keycode(layer_now,(keypos_t){.row = 2,.col = 6});
		uint16_t encoder_caw_keycode        = keymap_key_to_keycode(layer_now,(keypos_t){.row = 3,.col = 5});
		uint16_t encoder_cacw_keycode       = keymap_key_to_keycode(layer_now,(keypos_t){.row = 3,.col = 6});
		uint16_t encoder_cbw_keycode        = keymap_key_to_keycode(layer_now,(keypos_t){.row = 5,.col = 0});
		uint16_t encoder_cbcw_keycode       = keymap_key_to_keycode(layer_now,(keypos_t){.row = 5,.col = 1});
		uint16_t encoder_cdw_keycode        = keymap_key_to_keycode(layer_now,(keypos_t){.row = 5,.col = 2});
		uint16_t encoder_cdcw_keycode       = keymap_key_to_keycode(layer_now,(keypos_t){.row = 5,.col = 3});
    if (index == 0) { /* 编码器1 */
         if (clockwise) {
            tap_code(encoder_cw_keycode);
        } else {
            tap_code(encoder_ccw_keycode);
        }
    } else if (index == 1) { /* 编码器2 */
         if (clockwise) {
            tap_code(encoder_caw_keycode);
        } else {
            tap_code(encoder_cacw_keycode);
        }
    }else if (index == 2) { /* 编码器3 */
         if (clockwise) {
            tap_code(encoder_cbw_keycode);
        } else {
            tap_code(encoder_cbcw_keycode);
        }
    }else if (index == 3) { /* 编码器4 */
         if (clockwise) {
            tap_code(encoder_cdw_keycode);
        } else {
            tap_code(encoder_cdcw_keycode);
        }
    }
        return true;
}

xyzz, 2022-02-10:
Please upload the entire keyboard configuration including the `vial` keymap.

closed 2022-03-30 by xyzz