vial-kb/vial-qmk #123: Documentation clarification

exhuma, 2022-02-16:
How does vial compare to via? I can see a guide for "porting via to vial" but it does not state any potential limitations or risks.

I am asking this because I bought a PCB which states that it supports QMK but is *actually* built with VIA in mind. QMK sort of works until you enable backlighting and add more layers. I almost bricked my keyboard.

I came across "vial" and it gives me hope that it might be an interesting alternative to VIA (primarily because VIA is pretty closed source).

Are there any known risks and/or limitations known when running "vial" on a board that was developed for "via"?

Additionally, how far can i go with customisation on "vial". I would like to have a key that disables backlighting on all keys *except* the WASD cluster. This is currently not possible with "via" but *might* be doable with QMK. So how about "vial"?

BenRoe, 2022-02-22:
VIAL and VIA is built on top of QMK. So there is no "is actually built with VIA in mind".
If you keyboard does not work with some features and QMK, maybe there is some miss-configuration.
It would also help if you tell us what keyboard you have.

If a keyboard works with QMK, then it's possible to port it to VIA, or VIAL. 

To your question. Biggest difference is that VIAL is open source and actively developed. It has also support for rotary encoder.
VIA is closed source and latest release is from  19 Jun 2020.

exhuma, 2022-02-27:
Thank you for the reply. I do have a DZ60RGB from KBDFans which I only recently bought. I tried to flash it with QMK but then the RGB lighting failed (everything else was OK). Upon contacting the KBDFans support they stated that their latest revision is only working with VIA. And when I came across VIAL I was hoping that it could be an alternative. But I'm a bit wary and don't want to brick my keyboard by accident.

closed 2022-05-04 by xyzz