vial-kb/vial-qmk #159: [Bug] linkage error

EugenePY, 2022-05-04:
## Describe the Bug
Hi, I found linkage error during creating a pull request for my board(akira-60), and I encounter the same error with other board using the same chip(stm32f042x), the linkage error seems saying there are no enough RAM to fit on the board, but I can compile the board successfully on the qmk/qmk_firmware master.

**Linkage Error of the existing board using stm32f042x**
```shell
Linking: .build/chavdai40_rev1_default.elf                                                          [ERRORS]
 |
 | /usr/local/Cellar/arm-gcc-bin@8/8-2019-q3-update_1/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld:rules_memory.ld:314 cannot move location counter backwards (from 0000000020001da8 to 0000000020001800)
 | collect2: error: ld returned 1 exit status
```
**Compliation Result from qmk/qmk_firmware  master**
```shell
Creating binary load file for flashing: .build/ekow_akira_default.bin                               [OK]
Creating load file for flashing: .build/ekow_akira_default.hex                                      [OK]

Size after:
   text    data     bss     dec     hex filename
      0   21898       0   21898    558a ekow_akira_default.bin
```

## System Information

**Keyboard**: chavdai40/rev1
**Revision (if applicable):** 
**Operating system:** 
**`qmk doctor` output:** 
```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.0.0
Ψ QMK home: /Users/bigtreehouse/model_routine/keyboard/vial-qmk
Ψ Detected macOS 11.4.
⚠ QMK home does not appear to be a Git repository! (no .git folder)
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.3.1
Ψ Found avr-gcc version 8.4.0
Ψ Found avrdude version 6.3
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ QMK is ready to go, but minor problems were found
```

**Any keyboard related software installed?** 
 - [ ] AutoHotKey (Windows)
 - [ ] Karabiner (macOS)
 - [ ] Other:

## Additional Context

[](url)

zykrah, 2022-05-04:
This might be related to the difference between the two following:
https://github.com/vial-kb/vial-qmk/blob/vial/platforms/chibios/eeprom_stm32_defs.h#L26
https://github.com/qmk/qmk_firmware/blob/master/platforms/chibios/eeprom_stm32_defs.h#L26

zykrah, 2022-05-04:
> This might be related to the difference between the two following: https://github.com/vial-kb/vial-qmk/blob/vial/platforms/chibios/eeprom_stm32_defs.h#L26 https://github.com/qmk/qmk_firmware/blob/master/platforms/chibios/eeprom_stm32_defs.h#L26

In my testing, I can confirm this is what causes the compilation issue.

`FEE_PAGE_COUNT 8`
[image]

`FEE_PAGE_COUNT 2`
[image]

xyzz, 2022-05-04:
The eeprom amount in vial-qmk is increased to support vial features, you can reduce it in your keyboard config by defining `FEE_PAGE_COUNT`.

closed 2022-05-04 by xyzz