vial-kb/vial-qmk #215: tap-hold behavior

jhorology, 2022-08-01:
Is following behavior of tap-hold is correct?
```
E.g.) normaly alt, shift on tap_hold.

      |<--tapping term-->|
alt __|‾|___|‾‾‾‾‾‾‾‾‾‾‾‾‾‾‾‾‾‾‾‾‾ 
'a' ______________|‾|________|‾|___
                  'a'        'A'
```
If it's not the intention, I think it's better and natural behavior to fix [this part](https://github.com/vial-kb/vial-qmk/blob/874e867764716c1182690f7607fe1305cf06da0a/quantum/vial.c#L347-L349) as follows:

```c
        if (state->pressed) return DOUBLE_HOLD;
        else if (state->interrupted) return DOUBLE_SINGLE_TAP;
        else return DOUBLE_TAP;

```

xyzz, 2022-08-01:
Yes, it is intended to mimic qmk's behavior: https://docs.qmk.fm/#/feature_tap_dance

>  * Interrupted: If the state of a dance is "interrupted", that means that another key has been hit
>  *  under the tapping term. This is typically indicitive that you are trying to "tap" the key.

jhorology, 2022-08-01:
Is there a reason for following just a example code? I can't imagine advantage and use case of DOUBLE_SINGLE_TAP. Beside, VIAL can't define keycode of DOUBLE_SINGLE_TAP.

xyzz, 2022-08-02:
It's as explained in the docs -

>        // Last case is for fast typing. Assuming your key is `f`:
>        // For example, when typing the word `buffer`, and you want to make sure that you send `ff` and not `Esc`.
>        // In order to type `ff` when typing fast, the next character will have to be hit within the `TAPPING_TERM`, which by default is 200ms.

Yes you cannot define a separate key but it will enter the "tap" key twice in this case.