vial-kb/vial-qmk #275: Underglow Brightness and the Value of Underglow Color.

chronocide-commando, 2022-11-02:
Setting the Value (as in HSV) of Underglow Color currently sets the Underglow Brightness for the entire keyboard (and vice versa).

This behavior prevents the user from setting different brightness/Value for individual LEDs using Lighting Layers, instead forcing the same value for the entire keyboard.

Suggestion: Underglow Brightness should work as an additional multiplier applied to LED brightness/Value instead of setting it directly. This way it would control the overall brightness while preserving the relative difference between LEDs.

TweetyDaBird, 2024-03-01:
This is in-fact the desired and expected behavior. Underglow per definition is intended to affect all of the RGBs. 

To use the RGB diodes individually you should assign them as an RGB matrix first and use those functions. That gives you a lot more granular control.

chronocide-commando, 2024-03-04:
RGB Matrix is for keyboards with external drivers, if I understand correctly (which my keyboard doesn't have), but both RGB Light and RGB Matrix allow for individual LED control. RGB Light has [Lighting Layers](https://github.com/qmk/qmk_firmware/blob/master/docs/feature_rgblight.md#lighting-layers) and RGB Matrix has [Indicators](https://github.com/qmk/qmk_firmware/blob/master/docs/feature_rgb_matrix.md#indicators), but they do the same thing and are apparently intended for the same purpose.

For the LEDs individually controlled through Lighting Layers it would make more sense for the Underglow Brightness setting to affect them like a Brightness setting on TVs, by _proportionately_ changing the brightness of every pixel rather than setting them all to have the same brightness.

Right now the Val of Lighting Layers is either directly set to exactly the same number as Underglow Brightness (while Hue and Sat retain their values set by user) or completely unaffected by it, depending on RGBLIGHT_LAYERS_RETAIN_VAL.

TweetyDaBird, 2024-03-05:
RGB Matrix does not in any way require external drivers. It works fine with addressable RGB's as long as you define them as a matrix