vial-kb/vial-qmk #285: [BUG] Locked keyboard still allows for reading Matrix, editing keys, by VIA

JacobLChrzanowski, 2022-11-14:
A locked keyboard does not offer many of the immediately assumable security benefits of what seems to be advertised. When you come into VIAL and hear that you can lock your keyboard from modifications, that sounds great and dandy. My primary concern is silent matrix readout by an application that can talk to the keyboard over RAWHID. In reality this could extend to anything including Slack talking to your keyboard, and reading out its matrix. From reading out a keyboards matrix, you can start to assume some key placement, and start extracting messages, passwords, etc.

## Describe the Bug

When VIAL keyboards are "locked" the matrix is still able to be read out, macros are able to be deleted, key maps are able to be changed.
This exposes you to silent keylogging from applications, and breaking keyboard functionality by a program that decides to overwrite all your layers.
Perhaps this has not been done in practice but nothing is stopping a program from doing something like this, and having the lock function actually lock the keyboard configuration would make a lot of sense.

## Additional Context

What locking a keyboard should do, or at least allow us to toggle as an option, is restrict the matrix being read out to the pc to output only the keys defined in VIAL_UNLOCK_COMBO_ROWS and VIAL_UNLOCK_COMBO_COLS. While the keyboard is locked VIAL should also prevent writing to the keymap, or deleting macros.

You may say "but vial cannot read the matrix when the keyboard is locked"
Sure. But via can. https://usevia.app/#/

xyzz, 2022-11-14:
Can you confirm what firmware you are using on the keyboard, that sounds like a firmware with VIAL_INSECURE enabled?

JacobLChrzanowski, 2022-11-14:
I'm using Vial with below lines added, as well as having added an vial.json into the via folder to support compilation for Vial.
/keymaps/via/config.h
```
#define VIAL_KEYBOARD_UID {<random UID>}
#define VIAL_UNLOCK_COMBO_ROWS { 0, 2 }
#define VIAL_UNLOCK_COMBO_COLS { 0, 13 }
```
/keymaps/via/rules.mk
```
VIA_ENABLE = yes
VIAL_ENABLE = yes
```
/v2_1/rules.mk
```
VIAL_INSECURE = no
```

Vial.exe and the Vial website can lock and unlock my keyboard, and I am presented with the Esc+Enter prompt where I need to hold them down.
The firmware is whatever is newest for Vial in the vial-qmk repo (this repo).
I build with `qmk compile -kb dztech/dz60rgb/v2_1 -km via`
But this via keymap is not the stock via keymap. I made modifications on it to support compiling for Vial.
```
jacob:~/vial-qmk$ git status
On branch vial
Your branch is up to date with 'origin/vial'.
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
	modified:   keyboards/dztech/dz60rgb/keymaps/via/config.h
	modified:   keyboards/dztech/dz60rgb/keymaps/via/keymap.c
	modified:   keyboards/dztech/dz60rgb/keymaps/via/rules.mk
	modified:   keyboards/dztech/dz60rgb/v2_1/config.h
	modified:   keyboards/dztech/dz60rgb/v2_1/rules.mk
Untracked files:
  (use "git add <file>..." to include in what will be committed)
	keyboards/dztech/dz60rgb/keymaps/via/vial.json
no changes added to commit (use "git add" and/or "git commit -a")
```