vial-kb/vial-qmk #318: [Feature Request] DOIO/KB16 OLED always on option, Shiftkey Lock, and Alt+Gui keys in Quantum

antongeny, 2023-01-07:
#### DOIO KB16 Triple Knob Macro Pad feature request: 

Is there a way to always have the OLED screen always on? Option to adjust timeouts or toggle the screen would be helpful.

Is there a way to have the OLED screen to always be on? Or at least an option to have a timeout or always on? 

#### The keyboard I'm asking about: 

https://github.com/vial-kb/vial-qmk/tree/vial/keyboards/doio/kb16

#### QMK request: 

add Alt+Gui pairs(left and right versions) in quantum and toggle shift lock key (like LShift and Rshift keys). 

Alt+Gui combo keys don't exist. Had to customize those keys to function the way I needed.

#### Reason why I why a shift key toggle lock:

I'm utilizing the knobs on my DOIO KB 16 keyboard to navigate to specific areas in text. Then I'd highlight the text with shift, so I can cut/copy/paste whatever it is. Originally I made it so that it'd always highlight with shift+arrow keys, but I don't always need things to be highlighted. Would be super convenient if there was a shift toggle function for this reason.  

Regarding the shift key toggle, I've looked up and tried for hours in different apps to get the shift key to resemble what I need. Basically I need the Shift key to be press forever, until I press it again to turn it off. 

Great job on all the work that has been done. Thank you! 

## Feature Request Type

- [ ] Core functionality
- [X ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [x] New behavior

## Description

xyzz, 2023-01-07:
Please use https://feedback.vial.today/ for feature requests

closed 2023-01-07 by xyzz

antongeny, 2023-01-10:
> Alt+Gui combo keys don't exist. Had to customize those keys to function the way I needed

Thank you, I'll utilize that.