vial-kb/vial-qmk #322: [Bug] Build for RP2040 based boards fails

ijskegel, 2023-01-12:
## Describe the Bug

Build the code for a 3x3 macropad that uses a raspberry pi pico as MCU fails. To make sure it was not caused by my code I also tried to build vial_example/vial_rp2040:vial but that fails with the same errors.

The following error shows up several times:

```
./lib/pico-sdk/src/rp2_common/pico_bootrom/include/pico/bootrom.h:120:59: error: array subscript 0 is outside array bounds of 'uint16_t[0]' {aka 'short unsigned int[]'} [-Werror=array-bounds]
  120 | #define rom_hword_as_ptr(rom_address) (void *)(uintptr_t)(*(uint16_t *)rom_address)
      |                                                          ~^~~~~~~~~~~~~~~~~~~~~~~~~
./lib/pico-sdk/src/rp2_common/pico_bootrom/include/pico/bootrom.h:130:41: note: in expansion of macro 'rom_hword_as_ptr'
  130 |     uint16_t *func_table = (uint16_t *) rom_hword_as_ptr(0x14);
      |                                         ^~~~~~~~~~~~~~~~
```
## System Information

**Keyboard:** vial_example/vial_rp2040:vial
**Revision (if applicable):** -
**Operating system:** linux
**`qmk doctor` output:** 
```
~/repos/vial-qmk (ijskegel =) > qmk doctor
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: /home/patrick/repos/vial-qmk
Ψ Detected Linux.
Ψ Git branch: ijskegel
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.2.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.0
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2022-06-11 09:09:45 +0000 --  (f836d24b0)
Ψ - lib/chibios-contrib: 2022-07-27 10:46:15 +0200 --  (d03aa9cc)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/pico-sdk: 2022-05-17 19:19:01 +0200 --  (07edde8)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ QMK is ready to go, but minor problems were found
```

**Any keyboard related software installed?** 
no

## Additional Context

Updating to the same lib/pico-sdk submodule version as used in qmk solves the issue. 
This works for me:
```
cd lib/pico-sdk
git checkout 8d56ea332b3734cef0a8e61f7d61f2422bd539b1
cd ../..
make vial_example/vial_rp2040:vial
```

ijskegel, 2023-01-15:
Ok, thanks. I will continue with the workaround untill changes from QMK are pulled in.

closed 2023-01-15 by ijskegel

piit79, 2023-05-22:
Hey, did the fixes from QMK still not make it into Vial...? I'm still experiencing the same issue...