vial-kb/vial-qmk #352: [Bug] Not able to flash another firmware with security on

Crandel, 2023-02-01:
First time I flash vial firmware, compiled with default settings, which include security feature. Now I played with my custom keyboard settings and try to flash it again via qmk, but controller is not writable, even if I boot to bootloader via vial application.
I use crkbd/rev1 split keyboard and this setup require to flash firmware for both splits.
I need more detailed instruction for split keyboards 

## Describe the Bug

```bash
$ qmk flash -kb crkbd/rev1 -km crandel
Ψ Compiling keymap with make --jobs=1 crkbd/rev1:crandel:flash

Making crkbd/rev1 with keymap crandel and target flash

avr-gcc (GCC) 12.2.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text    data     bss     dec     hex filename
      0   26098       0   26098    65f2 crkbd_rev1_crandel.hex

Compiling: quantum/via.c                                                                            [OK]
Linking: .build/crkbd_rev1_crandel.elf                                                              [OK]
Creating load file for flashing: .build/crkbd_rev1_crandel.hex                                      [OK]
Copying crkbd_rev1_crandel.hex to qmk_firmware folder                                               [OK]
Checking file size of crkbd_rev1_crandel.hex                                                        [OK]
 * The firmware size is fine - 26098/28672 (91%, 2574 bytes free)
Waiting for USB serial port - reset your controller now (Ctrl+C to cancel)...................................
Device /dev/ttyACM1 has appeared; assuming it is the controller.
Waiting for /dev/ttyACM1 to become writable....................
```

## System Information

**Keyboard: crkbd** 
**Revision (if applicable): rev1** 
**Operating system: Arch Linux** 
**`qmk doctor` output:** 
```bash
$ qmk doctor
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: /data/vial-qmk
Ψ Detected Linux.
Ψ Git branch: vial
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.2.0
Ψ Found avr-gcc version 12.2.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 7.0
Ψ Found dfu-util version 0.11
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2022-06-11 09:09:45 +0000 --  (f836d24b0)
Ψ - lib/chibios-contrib: 2022-07-27 10:46:15 +0200 --  (d03aa9cc)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/pico-sdk: 2022-05-17 19:19:01 +0200 --  (07edde8)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ QMK is ready to go, but minor problems were found
```

**Any keyboard related software installed?** 
 - [ ] AutoHotKey (Windows)
 - [ ] Karabiner (macOS)
 - [ ] Other:

## Additional Context

Crandel, 2023-02-15:
It was vial problem. My hardware reset button was not working even after security unlock. The issue was solved only after making Reset buttons in default layout on both splits as they should be flashed with the same firmware separately

TweetyDaBird, 2023-04-23:
> The main issue is not hardware soldering, as this key is working perfectly after disabling vial security. The main issue is to enable security by default without the instruction how to flash correctly again.

What you are describing is simply not possible. A physical reset button on the keyboard/controller is read by the bootloader of the MCU, putting it in DFU/flashing mode before any code such as the Vial firmware is even loaded into memory and run. 

Thus, Vial simply cannot interfere with the _physical_ reset button.

The whole point of the locking is that it only prevents someone without _physical_ access to the keyboard from flashing. All ways to reset the controller that requires physical access circumvents the lock by default.

Crandel, 2023-04-23:
As I said, the only way to boot into bootloader is to Add Reset button into main layout for both sides. Reset buttons were not working with security enabled.

closed 2023-04-23 by Crandel

TweetyDaBird, 2023-04-23:
> As I said, the only way to boot into bootloader is to Add Reset button into main layout for both sides. Reset buttons were not working with security enabled.

Again, this is electrically impossible. The _physical_ reset switch is tied to an actual IO pin, and controlled by the bootloader, not the firmware. The two are separate, and the bootloader is what starts up and loads the firmware.

If the reset pin is held to ground by the switch, the bootloader never loads the firmware (which contains the locking part) and it cannot have any effect on the bootloader and how it works. Simply because it hasn't been loaded or run yet at that point. 

It's like saying you re-programmed your door lock and suddenly you cannot put the key into the lock. One part is programming, the other is a physical (an electrical connection in this case) and cannot be affected by the programming.