vial-kb/vial-qmk #384: [Feature Request] Apply Joystick improvements from upstream qmk

9R, 2023-03-01:
## Feature Request Type

- [x] Core functionality
- [x] Improve hardware support: Joystick
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

QMK merged some improvements for the joystick implementation in January:  qmk/qmk_firmware#19052

This changed some variables which makes it harder to maintain configurations for qmk and qmk-vial.

Because of these changes following current QMK documentation on how to implement joysticks does not work with Vial any more and will result in failed builds and there is no Vial-specific documentation describing the current implementation in vial. This currently makes it harder to port new devices with Joysticks or analog inputs to Vial.

Here is a summary of improvements copied from the original pull requests:
   - Improved documentation, added API reference
   - Renamed JOYSTICK_AXES_COUNT/JOYSTICK_AXES_RESOLUTION to singular AXIS to align with JOYSTICK_BUTTON_COUNT
   - Replaced enum joystick_status with simple dirty flag in joystick_t
   - Added V-USB support
   - Added shared EP support for LUFA/ChibiOS

Is it possible to merge these changes in Vial?

Cujo26, 2023-03-20:
Vial periodically merges changes from upstream, so new features should get implemented, just on a bit of a delay compared to the main project.

9R, 2024-10-05:
Sorry for the long response time on my end. I have tested against current vial and the issue is not present any more.

I have also created a [PR](https://github.com/vial-kb/vial-qmk/pull/798) for the keyboard that is using this feature.

closed 2024-10-05 by 9R