vial-kb/vial-qmk #437: [Bug]

1ightmagic, 2023-04-04:
## Describe the Bug

## System Information

**Keyboard:** 
**Revision (if applicable):** 
**Operating system:** 
**`qmk doctor` output:** 
```
(Paste output here)
```

**Any keyboard related software installed?** 
 - [ ] AutoHotKey (Windows)
 - [ ] Karabiner (macOS)
 - [ ] Other:

## Additional Context

I made a homemade ATmega32U4 MCU keyboard and recently added a CAT24C512 EEPROM ( used with previous OLEDs on the I2C bus ) to expand the functionality of Macros.

But after rebuilding the firmware, it was found that the keyboard was unusable.

After multiple attempts, I found that OLED and eeprom functions seem to be still normal,

But the preset key layers and values will switch quickly and randomly, and the RGB function will no longer be available.

At this point, if the key value is pressed, the entire keyboard will immediately hang without any response.

I used the exclusion method to uninstall all the separable functions one by one, and finally found out that it was Vial's problem.

Vial caused the keyboard to go crazy after enabling Combos and Key Overrides functions.

After completely disabling these two functions, eeprom, oled, RGB, and the remaining button functions can all be used normally.

paprikman, 2023-04-06:
> As noted on the [firmware size reduction page](https://get.vial.today/docs/firmware-size.html), Vial's additional features consume both EEPROM and RAM. RAM exhaustion with Vial firmware containing many features is pretty common on atmega32u4...even before adding discrete EEPROM to try and cram more things in.
> 
> Switching to a better MCU would be a good idea if you require all of these features.

So this RAM exhaustion can be a reason why my atmega 32u4 board resets itself to the stock (firmware defined) settings on occasion?

1ightmagic, 2023-04-07:
> As noted on the [firmware size reduction page](https://get.vial.today/docs/firmware-size.html), Vial's additional features consume both EEPROM and RAM. RAM exhaustion with Vial firmware containing [...]

Maybe? These two days, I tried to borrow a sample of STM32F303CBT6. This phenomenon no longer exists. It is worth noting that STM32F303 also has some error prompts on Vial's control panel, similar to ATmega32U4, after each firmware refresh, but it does not affect usage.

1ightmagic, 2023-04-07:
> As noted on the [firmware size reduction page](https://get.vial.today/docs/firmware-size.html), Vial's additional features consume both EEPROM and RAM. RAM exhaustion with Vial firmware containing [...]

> Maybe? These two days, I tried to borrow a sample of STM32F303CBT6. This phenomenon no longer exists. It is worth noting that STM32F303 also has some error prompts on Vial's control panel, similar [...]

By the way, please ask. Is there a method in QMK that can directly write firmware to external eeprom? This way, as long as the MCU is initially started, it is sufficient to directly read and use the external eeprom. It can avoid the problem of outdated MCU storage being too small or with inconsistent specifications, allowing for a certain degree of liberation for multiple types of MCU. And you can confidently put more functions into operation. Think further and even use storage cards? Hard drive? SSD？ Hahaha, maybe Vial can do this. Of course, the premise is that safety comes first.

TweetyDaBird, 2023-04-23:
> By the way, please ask. Is there a method in QMK that can directly write firmware to external eeprom? This way, as long as the MCU is initially started, it is sufficient to directly read and use the [...]

No such way to use ee_prom for the main firmware exist, and I don't believe it will ever be a priority to create it. QMK as such is slowly and quietly phasing out older MCU's like the Atmega32u4 by simply overwhelming it's specs with new and added functions. So I find it very unlikely that they should create a slightly less than cost effective way to keep those alive in the ecosystem. 

And since Vial strives to maintain backward compatibility with QMK as much as possible, it's not likely a feature that will get added that way either.

Ardakilic, 2023-07-30:
We can reproduce this incident on RP2040.

Please see the recent actions:

https://github.com/Ardakilic/Skywatch-QMK

https://github.com/Ardakilic/Skywatch-QMK/actions/runs/5708733964/job/15466684407

Oled + Vial + Key combos are present on the board.

The keymap file builds successfully on qmk_firmware repository, but gives issues on vial.

https://github.com/Ardakilic/Skywatch-QMK/blob/master/skywatch/keymaps/arda/keymap.c

TweetyDaBird, 2023-07-31:
I kind of doubt they are related. What you have there is a plain and simple mess. 

It might **compile** in QMK, but it should never make it into repo, as there are piles of things in that file that should have been moved out to other more high level files, simplifying things, such as the OLED definitions etc.

#And the combination of hardware, or possible ram exhaustion have nothing to do with your issue, it's just a plain compile error, which you'd have to work through and find the reason for. I have compiled multiple times with vial and using OLEDs/RGB etc combined with rp2040 and an ee_prom (since they require one). 

These are **NOT** related, seek the error in your code.

closed 2023-07-31 by xyzz