vial-kb/vial-qmk #516: Add info.json for libra mini

weensy, 2023-08-06:
I bought a libra mini, but "no devices detected" comes out from vial.
My libra mini has different keyboard_name, vid, pid, device_version.
Can you create an info.json for this?
[image]

weensy, 2023-08-26:
@lesshonor Thanks for the reply, but I solved this problem without reflash.

For those who have the same problem as me, refer to the repository below.
https://github.com/KobeCofini/LibraMini

TweetyDaBird, 2024-03-01:
> @lesshonor Thanks for the reply, but I solved this problem without reflash.
> 
> 
> 
> For those who have the same problem as me, refer to the repository below.
> 
> https://github.com/KobeCofini/LibraMini

Please don't. That's side loading a VIA firmware to an old outdated Vial version.