vial-kb/vial-qmk #522: QMK settings not working

linkpy, 2023-08-12:
using keymap : https://github.com/linkpy/vial-qmk/tree/vial/keyboards/splitkb/aurora/sofle_v2/keymaps/vial

with qmk settings enabled, changing tapping term doesnt do anything (even setting it to a very high value like 1000ms doesnt seem to have any impact on the actual tap-hold behaviours).

i only tested this with LT_n keys.

to reproduice : 
- enable qmk settings 
- flash
- try to change the tapping term with the vial app
- notice nothing changes regarding key rolls and LT_n keys

stefanheyder, 2023-08-31:
I am encountering the same issue which make using home row mods with QMK keys unusable unfortunately ...

xyzz, 2023-09-03:
I couldn't reproduce the issue with LT_n, please provide better/more detailed reproduction steps.

XavRan, 2023-12-05:
Facing the same problem on default cantor keymap [https://github.com/linkpy/vial-qmk/tree/vial/keyboards/cantor](https://github.com/linkpy/vial-qmk/tree/vial/keyboards/cantor) I am able to change the tapping term from QMK settings, but rolling 2 keys within the tapping term simply activates the modifier. Just like permissive hold. Even though it is disabled.

xyzz, 2024-01-21:
I'm closing this issue because no reproduction steps were provided and we cannot act on this issue.

> Facing the same problem on default cantor keymap

I don't see any modifier keys on the linked keymap so I'm not sure how you are running into the same issue.

closed 2024-01-21 by xyzz