vial-kb/vial-qmk #622: Drop CTRL V1, Vial keymap flash throws "Unsupported protocol version!" error when launching vial

RPINerd, 2023-12-17:
I flashed my original Ctrl from drop with the firmware in this repo under keyboards/massdrop/ctrl and with ./keymap/vial
When I launch vial I get a popup that says:

> Unsupported protocol version!
> Please download latest Vial from https://get.vial.today/

I did the exact same thing with The Key V2 from that same folder and have no issues with vial seeing it

Also I'm not sure if this is a problem to open here, or with vial-gui?

AndrewJRichardson, 2023-12-22:
Are you compiling from this repo or getting a precompiled firmware from somewhere else?

It seems you can use Drop's own configurator (the downloaded version) to get a Vial compatible fireware) https://drop.com/talk/9382/how-to-configure-your-ctrl-keyboard.

RPINerd, 2023-12-22:
I'm compiling and installing from this repo, from the two paths I listed in the op:

./keyboards/massdrop/ctrl/
./keyboards/massdrop/ctrl/keymap/vial/

rmblau, 2023-12-28:
Hi,

How are you compiling the firmware? I'm running this firmware on my v1 without any issues.

Thanks!

RPINerd, 2023-12-28:
Using the vial qmk repo:

```bash
qmk config user.keyboard=massdrop/ctrl
qmk compile -km massdrop/ctrl/keymaps/vial/
```

And then flashing also using the same method.. If it's working for you then clearly I have something wrong with my following of the wiki lol

rmblau, 2023-12-28:
> Using the vial qmk repo:
> 
> ```bash
> qmk config user.keyboard=massdrop/ctrl
> qmk compile -km massdrop/ctrl/keymaps/vial/
> ```
> 
> And then flashing also using the same method.. If it's working for you then clearly I have something wrong with my following of the wiki lol

Have you tried the make command or running QMK compile from the massdrop directory? I originally ported Vial to this keyboard and believe I've been running it with make.

rmblau, 2023-12-28:
Looking through the wiki if you use the make command you need to be running it from the root of vial-qmk with `make keyboard/keyboard_model:vial`

RPINerd, 2023-12-28:
Thanks @rmblau looks like using make instead of qmk compile was the way to go!

For anyone in the future, this was the full command process:

```bash
cd /home/[me]/qmk-vial/
make massdrop/ctrl:vial:flash
```
And then activating flash mode when prompted by holding Fn+b for 3 seconds

Cheers!

closed 2023-12-28 by RPINerd