vial-kb/vial-qmk #646: .gitignore should not ignore keymap's info.json

Jari27, 2024-01-09:
The `info.json` file in QMK can be used to provide a [data-driven](https://docs.qmk.fm/#/data_driven_config) definition of `keyboard.c` and `rules.mk`. Is there any reason [this commit](https://github.com/vial-kb/vial-qmk/commit/38700f7e3fb6806a263c95d39aa1719d8146288c) added it to `.gitignore`?

Does vial not support `info.json` (completely)? 

Otherwise, I think it's valuable to allow users to define (most of) their keymaps using  `info.json` instead of `keymap.c` and `rules.mk`.

Jari27, 2024-01-09:
Tagging the author: @zvecr

zvecr, 2024-01-10:
> .gitignore should not ignore keymap's info.json

This is conceptually incorrect.

`info.json` has never been supported at the keymap level. Updating the `.gitignore` attempts to improve PR quallity by both preventing users from committing the file in the first place, and flagging the addition in CI.

> Otherwise, I think it's valuable to allow users to define (most of) their keymaps using info.json instead of keymap.c and rules.mk.

`keymap.json` is the data-driven equivalent of `keymap.c`. The current issue is that the 2 files are mutually exclusive.

In the future there is a plan to provide a more cohesive way to support mixed data-driven and traditional config.

> Does vial not support info.json (completely)?

Not the `qmk/qmk_firmware` repo so i’m afraid I can’t assist with that question :shrug:.

Jari27, 2024-01-10:
> [qmk/qmk_firmware#20408](https://github.com/qmk/qmk_firmware/pull/20408)
> 
> Please don't tag QMK maintainers in the Vial repo.

My bad, didn't realize the original commit came from QMK.