vial-kb/vial-qmk #650: Nuphy keyboard

MoroJr, 2024-01-17:
Can you add support for Nuphy Air 75 v2?

https://nuphy.com/products/air75-v2

adophoxia, 2024-01-18:
@MoroJr, to copy this from the issue template for the `qmk/qmk_firmware` as this is also relevant:
> Please refrain from asking maintainers to add support for specific keyboards -- it is unlikely they will have hardware available, and will not be able to help.
> Your best bet is to take the initiative, add support, then submit a PR yourself.

And considering how the PR to add this board in `qmk/qmk_firmware` is going, it might be some time before this'll happen. To add to that, historically, in order for a board to be merged into `vial-kb/vial/qmk`, it has to be first merged into `qmk/qmk_firmware` so as to reduce/not deal with any merge conflicts when the Vial fork merges from upstream QMK.

closed 2024-01-21 by xyzz

roryokane, 2024-04-17:
For reference, upstream QMK’s pull request: https://github.com/qmk/qmk_firmware/pull/22751