vial-kb/vial-qmk #654: Drop Ctrl V1 doesn't work properly

endigma, 2024-01-20:
After flashing the massdrop/ctrl:vial firmware through the make commands the keyboard shows up in VIAL but has wonky mappings and layers don't work, neither does bootloader reboot even if on layer 1.

Wonky mappings:
[image]

I can't really show evidence of the bootloader not working or the layers not working but they do not.

rmblau, 2024-02-09:
Hi,

I'll look into this and see what I can come up with. I have no issues on my drop ctrl. Have to remind myself of any changes I've made that aren't in the keymap when I ported this. If I made any non keymap related changes that is.

elibakshi, 2024-12-13:
Hello,

I'm currently trying to create a vial port for my alt v1 based on your current ctrl v1 vial port, but I am running into the same issue described above. I was wondering you had any updates on what could be causing this behavior. Appreciate any help!

closed 2026-09-10 by endigma