vial-kb/vial-qmk #664: Tap Term does not work for LT(1)

tomwang2011, 2024-02-06:
I have been trying to put LT(1) on spacebar such that tapping spacebar gives space and holding spacebar goes to layer 1. 

According to QMK docs we should see something like: https://github.com/qmk/qmk_firmware/blob/master/docs/tap_hold.md#nested-tap-abba-idnested-tap where as long as I am within my tap term range, i press `space` then `A` gives me ` a`. However if I hold `space` then tap `A` I should get `left` (I configured Layer 1 wasd to be directions).

As I am getting more `left` instead of ` a`, I tried to increase the tap term under qmk settings. However even at an unrealistic 5000ms, I am still only getting `left` instead of ` a`, so I think `LT()` probably doesn't work with tap term?

I tried the same when flashing qmk and it works fine.

refact0r, 2024-03-16:
Try enabling and disabling `Permissive Hold` and `Ignore Mod Tap Hold Interrupt` and seeing if it changes anything.

pmympmm, 2024-04-24:
Hi, 
I'm new to vial (my old keyboard used qmk and i didnt see the same - i am trying to use the same layout) but seeing something very similar to Tom. Did you figure it out @tomwang2011  ??

Vial 0.7.1 on linux/ubuntu.
my keyboard is a keyclicks ergo

I would upload my config but it has a bunch of macros and things with my name and other personal bits.

I type quite quickly and my home row is usually LT keys but what I find is that I often end up with keys from layers scattered into what I type.

I have also made tapping_term much much higher and it seems to have zero effect. eg setting it to 2000 doesn't change anything. This seems repeatable with ctrl and layer modifiers.

i've tried with combinations of permissive hold & ignore mod tap interrupt but get the same.

sleeplesslord, 2024-05-13:
I'm having the same problem with the Keychron Q11. The behavior of any key configured using mod tap is as if Permissive Hold is enabled even though it's disabled, and updating the tapping term doesn't seem to do anything.

ntubiolin, 2024-06-29:
> ~I'm having the same problem with the Keychron Q11. The behavior of any key configured using mod tap is as if Permissive Hold is enabled even though it's disabled, and updating the tapping term doesn't seem to do anything.~ I tried enabling Ignore Mod Tap Interrupt and it seems to work the way I would expect now.

It works for me. Thanks!
Also found another explanation useful: https://reddit.com/r/olkb/comments/gmcb57/qmk_ignore_mod_tap_interrupt_documentation/

apprehensions, 2025-01-02:
I'm having this issue and nothing works for me. I switched to vial from via specifically to control the tap timeout, but it seems like its not functional.

apprehensions, 2025-01-02:
Turns out that for [303 keyboards](https://github.com/search?q=repo%3Avial-kb%2Fvial-qmk+NO_ACTION_TAPPING+path%3A%2F%5Ekeyboards%5C%2F%2F&type=code) this feature is completely disabled. Why? To save space? Why not save space through other means like (in my case) `NO_PRINT`?