vial-kb/vial-qmk #668: multiple definition of `key_combos'

andrewlphilpott, 2024-02-12:
Is there a way to get combos working in the default keymap? I'm trying to add one based on how it's shown in the [QMK docs](https://github.com/qmk/qmk_firmware/blob/master/docs/feature_combo.md#combos) (I'm just copy/pasting their first example into my `keymap.c` file).

When I try to compile I get `multiple definition of 'key_combos'`.

I do have `COMBO_ENABLE = yes` in my `rules.mk` but is there something else I need to do as well?

Full error message below:

```
Linking: .build/qazimodo_default.elf                                                                [ERRORS]
 | 
 | /usr/local/Cellar/arm-none-eabi-gcc/10.3-2021.07/gcc/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: .build/obj_qazimodo_default/quantum/vial.o:/Users/andrewphilpott/Repos/vial-qmk/quantum/vial.c:509: multiple definition of `key_combos'; .build/obj_qazimodo_default/quantum/keymap_introspection.o:/Users/andrewphilpott/Repos/vial-qmk/./keyboards/qazimodo/keymaps/default/keymap.c:57: first defined here
 | collect2: error: ld returned 1 exit status
 | 
gmake: *** [builddefs/common_rules.mk:283: .build/qazimodo_default.elf] Error 1
```

andrewlphilpott, 2024-02-12:
Also, should have put this in the original post but I can get combos to work through the Vial GUI. Just can't get it to compile if I try to add any in the `keymap.c`.

andrewlphilpott, 2024-02-12:
Okay after doing some experimenting, I'm able to get QMK to compile if I disable Vial. So I'm assuming that means adding default combos to the keymap is not supported with Vial.

xyzz, 2024-02-12:
Yes, manually defining combos is not supported in the "vial" keymap since it conflicts with Vial's dynamic combos.

closed 2024-02-12 by xyzz

Lefuneste83, 2024-09-10:
Yes you just need to get rid of the combo entries in your keymap.c, define a nice number of vial combo entries using

`#define VIAL_COMBO_ENTRIES xx `(my choice is 25) in your config.h at keymap level

Then flash your KB and launch Vial. Then go to the Combos tab and define your combos. 

[image]

Don't forget to save your KB settings for future flashing as these will be reset every time you update your FW!