vial-kb/vial-qmk #726: KeyError 'layouts'

coshmeo, 2024-04-30:
I am on macOS 14.4.1 with a boardsource equals 48, i've got my firmware compiled and loaded onto the board, which works fine, but i am not able to interact with it using Vial. When i connect they keyboard to vial, i get the following error message:

```
File "main_window.py", line 279, in on_devices_updated
  File "main_window.py", line 283, in on_device_selected
  File "autorefresh/autorefresh.py", line 79, in select_device
  File "vial_device.py", line 51, in open
  File "protocol/keyboard_comm.py", line 79, in reload
  File "protocol/keyboard_comm.py", line 153, in reload_layout

KeyError: 'layouts'
```

My end goal is to be able to program the LEDs with Vial in hopes that i don't have to code it all by hand. Any help is appreciated, thank you!

coshmeo, 2024-04-30:
my `vial.json` looks like this:
```
{
  "version": 1,
  "notes": "",
  "documentation": "\"This file is a QMK Configurator export. You can import this at <https://config.qmk.fm>. It can also be used directly with QMK's source code.\n\nTo setup your QMK environment check out the tutorial: <https://docs.qmk.fm/#/newbs>\n\nYou can convert this file to a keymap.c using this command: `qmk json2c {keymap}`\n\nYou can compile this keymap using this command: `qmk compile {keymap}`\"\n",
  "keyboard": "boardsource/equals/48",
  "keymap": "coshmeoside",
  "layout": "LAYOUT_ortho_4x12",
  "layers": [
    [
      "KC_GRV",
      "KC_Q",
      "KC_W",
      "KC_E",
      "KC_R",
      "KC_T",
      "KC_Y",
      "KC_U",
      "KC_I",
      "KC_O",
      "KC_P",
      "KC_BSPC",
      "KC_TAB",
      "KC_A",
      "KC_S",
      "KC_D",
      "KC_F",
      "KC_G",
      "KC_H",
      "KC_J",
      "KC_K",
      "KC_L",
      "KC_SCLN",
      "KC_QUOT",
      "KC_LSFT",
      "KC_Z",
      "KC_X",
      "KC_C",
      "KC_V",
      "KC_B",
      "KC_N",
      "KC_M",
      "KC_COMM",
      "KC_DOT",
      "KC_SLSH",
      "KC_ENT",
      "MO(1)",
      "KC_LCTL",
      "KC_LALT",
      "KC_LGUI",
      "MO(2)",
      "KC_CAPS",
      "KC_SPC",
      "KC_SPC",
      "KC_LEFT",
      "KC_DOWN",
      "KC_UP",
      "KC_RGHT"
    ],
    [
      "KC_TILD",
      "KC_EXLM",
      "KC_AT",
      "KC_HASH",
      "KC_DLR",
      "KC_PERC",
      "KC_CIRC",
      "KC_AMPR",
      "KC_ASTR",
      "KC_MINS",
      "KC_EQL",
      "KC_DEL",
      "KC_TAB",
      "KC_F1",
      "KC_F2",
      "KC_F3",
      "KC_F4",
      "KC_F5",
      "KC_F6",
      "KC_LPRN",
      "KC_RPRN",
      "KC_LBRC",
      "KC_RBRC",
      "KC_PIPE",
      "KC_TRNS",
      "KC_F7",
      "KC_F8",
      "KC_F9",
      "KC_F10",
      "KC_F11",
      "KC_F12",
      "KC_NO",
      "KC_NO",
      "KC_NO",
      "KC_BSLS",
      "KC_TRNS",
      "KC_NO",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_HOME",
      "KC_PGDN",
      "KC_PGUP",
      "KC_END"
    ],
    [
      "KC_ESC",
      "KC_1",
      "KC_2",
      "KC_3",
      "KC_4",
      "KC_5",
      "KC_6",
      "KC_7",
      "KC_8",
      "KC_9",
      "KC_0",
      "KC_BSPC",
      "KC_TAB",
      "KC_F1",
      "KC_F2",
      "KC_F3",
      "KC_F4",
      "KC_F5",
      "KC_F6",
      "KC_NO",
      "KC_MINS",
      "KC_EQL",
      "KC_LBRC",
      "KC_RBRC",
      "KC_TRNS",
      "KC_F7",
      "KC_F8",
      "KC_F9",
      "KC_F10",
      "KC_F11",
      "KC_F12",
      "KC_NO",
      "RGB_M_P",
      "RGB_RMOD",
      "RGB_MOD",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_LEFT",
      "KC_DOWN",
      "KC_UP",
      "KC_RGHT"
    ]
  ],
  "author": ""
}
```

coshmeo, 2024-05-01:
nvm, figured it out.

closed 2024-05-01 by coshmeo

GibbyWoyto, 2024-05-09:
Hi, 
I got the same issue and cant figure it out. yan you tell me what you did to solve it?

Thank You

coshmeo, 2024-05-09:
@GibbyWoyto i followed vial's instructions here: https://get.vial.today/docs/porting-to-via.html
but basically `vial.json` is just supposed to contain some basic info, not the whole keymap. 

it should look something like this (for an equals48):
```
{
    "lighting": "vialrgb",
    "matrix": {
        "rows": 4,
        "cols": 12
    },
    "layouts": {
        "keymap":[["0,0","0,1","0,2","0,3","0,4","0,5","0,6","0,7","0,8","0,9","0,10","0,11"],
        ["1,0","1,1","1,2","1,3","1,4","1,5","1,6","1,7","1,8","1,9","1,10","1,11"],
        ["2,0","2,1","2,2","2,3","2,4","2,5","2,6","2,7","2,8","2,9","2,10","2,11"],
        ["3,0","3,1","3,2","3,3","3,4","3,5","3,6","3,7","3,8","3,9","3,10","3,11"]]
    }
}
```

here is the vial folder i'm currently using - i still haven't figured out how to get per-key rgb working, but you can at least change the default patterns. feels like i'm pretty close to getting it, but haven't had time/motivation to work on it the last couple weeks. will probably get around to it eventually.
[vial.zip](https://github.com/vial-kb/vial-qmk/files/15267430/vial.zip)