vial-kb/vial-qmk #737: [feature] Keybinds for API triggers

gbdlin, 2024-05-22:
### Rationale
Currently, user is limited to use keys available in the standard keymap on every OS, which is limiting when implementing for example a macropad. Some may use additional key combinations, but they may often conflict with user holding a specific key down while using macropad + has other limitations when trying to combine macros together.

### Solution
Add a special keycode that will not trigger any keypress, but instead it'll output specific message to the rawhid endpoint that a specific key was pressed. Options for the exact message are:

- user-specified code or message, assigned to the key with the mentioned above keycode
- Position of the pressed key on the matrix + optionally current layer
- ...

### Security considerations
This keycode should be only assignable in unlocked mode, as attacker could replace all keybinds with this keycode to gain keylogging functionality.

xyzz, 2024-06-04:
please use https://feedback.vial.today for feature requests

closed 2024-06-04 by xyzz

gbdlin, 2024-06-04:
It would be nice to put it in some contribution guide in repository root, so it's easier to find