vial-kb/vial-qmk #766: Create Combo in Firmware

RickSeiden, 2024-08-04:
Hello,

I would like to put a combo into the firmware so it doesn't have to be added via the GUI every time a new version of the firmware is installed.

Use case: I've built a custom keyboard with a built-in trackball.  The trackball has no middle mouse button.  I'd like this built in combo to press KC_MS_BTN3 when KC_MS_BTN1 and CK_MS_BTN2 are pressed.

This works fine if I build the combo in the GUI.

When I try to build the combo into the firmware using the [instructions on QMK](https://docs.qmk.fm/features/combo) for building a combo and compile I get the following error:

> Compiling: quantum/keymap_introspection.c                                                           [OK]
> Compiling: quantum/via.c                                                                            [OK]
> Linking: .build/keyball_vial.elf                                                                    [ERRORS]
>  |
>  | /bin/../lib/gcc/avr/8.3.0/../../../../avr/bin/ld: .build/obj_keyball_vial/quantum/vial.o:/home/qmk/vial-qmk/quantum/vial.c:509: multiple definition of 'key_combos'; .build/obj_keyball_vial/quantum/keymap_introspection.o:/home/qmk/vial-qmk/./keyboards/keyball/keymaps/vial/keymap.c:20: first defined here
>  | collect2: error: ld returned 1 exit status
>  |
> gmake: *** [builddefs/common_rules.mk:283: .build/keyball_vial.elf] Error 1

My code for the combo is as follows:

```
const uint16_t PROGMEM middle_button[]={KC_MS_BTN1, KC_MS_BTN2, COMBO_END};

combo_t key_combos[]={
        COMBO(middle_button, KC_MS_BTN3)
};
```

The error says that key_combos is already defined, so I thought I'd remove the combo_t definition and just go with

```
key_combos[]={
        COMBO(middle_button, KC_MS_BTN3)
};
```

but that gave me an error that said

> Compiling: quantum/keymap_introspection.c                                                          In file included from quantum/keymap_introspection.c:5:
> ./keyboards/keyball/keymaps/vial/keymap.c:20:1: error: data definition has no type or storage class [-Werror]
>  key_combos[]={
>  ^~~~~~~~~~
> ./keyboards/keyball/keymaps/vial/keymap.c:20:1: error: type defaults to 'int' in declaration of 'key_combos' [-Werror=implicit-int]
> ./keyboards/keyball/keymaps/vial/keymap.c:21:2: error: braces around scalar initializer [-Werror]
>   COMBO(middle_button, KC_MS_BTN3)
>   ^~~~~

**Is there a way to include a combo in the firmware with Vial, and if so, how?**

Thanks

RickSeiden, 2024-09-06:
Nope. Nothing.

Lefuneste83, 2024-09-10:
This is a little annoying but Vial currently does not support Combos in keymap definitions when compiling.

You indeed need to remove them and define them in the GUI. 

But I would suggest that you make use of the Vial settings save feature in order to distribute a complementary Vial settings stock file (.vil file extension) with your keyboard. This is the way Vial expects you to handle your settings including Combos. I believe keymap settings should be kept at a minimum. QMK main repo admins also require default keymap to be super minimal, and I tend to believe it is about the same philosophy with Vial except that you have a super GUI to manipulate them!