vial-kb/vial-qmk #820: Permissive hold QMK setting doesn't apply to tap dance keys.

leg7, 2024-11-23:
Expected that holding my tap dance key and hitting an alpha key before the tapping term sends the modifier in the tap dance + the alpha key.

[image]
[image]
[layout2.vil.gz](https://github.com/user-attachments/files/17879578/layout2.vil.gz)

AntonioRodriguezRuiz, 2025-01-26:
I'm having the same problem over here. Did you ever found a solution for this?

leg7, 2025-01-30:
> I'm having the same problem over here. Did you ever found a solution for this?

Nope :(

getreuer, 2025-03-16:
**This is (for better or worse) working as intended. Permissive Hold does NOT affect Tap Dance.**

Internally, QMK has a core stage of event processing for mod-tap `MT` and layer-tap `LT` keys. It implements a fairly elaborate set of logic to decide whether the keys are "tapped" vs. "held." Permissive Hold (and the other options under the Tap-Hold tab) are options that modify this logic. See the [QMK tap-hold documentation](https://docs.qmk.fm/tap_hold) for detailed description.

Tap Dance, on the other hand, is implemented completely separately as a distinct stage of processing. Although Tap Dance also makes a decision about whether keys are "tapped" vs. "held," its logic is more basic. It does not reimplement Permissive Hold. The point of Tap Dance is to trigger different actions when the key is double tapped, etc., which is beyond the scope of the tap-hold implementation.

If you are looking to add home row mods to your keymap, I recommend using the tap-hold keys (`LSFT_T`, `RCTL_T`, ...) under the Quantum tab, not Tap Dance.

leg7, 2025-04-24:
> **This is (for better or worse) working as intended. Permissive Hold does NOT affect Tap Dance.**
>
> Internally, QMK has a core stage of event processing for mod-tap `MT` and layer-tap `LT` keys. It [...]

So it's impossible to use the config I posted?

getreuer, 2025-04-24:
@leg7 I'm afraid Vial is a dead end. It's "possible" to make that config. The issue is that this doesn't produce the desired behavior that you are after. Permissive Hold does not apply to Tap Dance. This is "working as intended" and not a QMK or Vial bug, for better or worse.

There is a solution if you switch to using QMK directly: There's still no out-of-the-box key for what you want, but it can be done with some custom handler code:

* Define an `MT` key `LCTL_T(KC_0)` ⇒ gets the desired hold function
* [Change the tap function](https://docs.qmk.fm/mod_tap#changing-tap-function) to behave as one-shot shift ⇒ on tap function
* Leverage [Quick Tap](https://docs.qmk.fm/tap_hold#quick-tap-term) to implement shift ⇒ on tap+hold function

That covers the three functions in the tap dance, but built as a tap-hold key, so it honors Permissive Hold. To toot my own horn, you can also use the new [Chordal Hold option](https://docs.qmk.fm/tap_hold#chordal-hold) (in QMK, but not exposed in Vial yet) for additional control.

Like this (tested):

```.c
// Copyright 2025 Google LLC.
// SPDX-License-Identifier: Apache-2.0

// Define the keycode. The "KC_0" is an arbitrary placeholder that is never sent.
#define CTL_SFT LCTL_T(KC_0)

// Use `CTL_SFT` somewhere in your layout...

bool process_record_user(uint16_t keycode, keyrecord_t* record) {
  switch (keycode) {
    // Implements:
    // * on tap:         one-shift Shift
    // * on hold:        Ctrl
    // * on tap + hold:  Shift
    case CTL_SFT:
      if (record->tap.count > 0) {
        if (record->tap.count == 1) {  // Single tap.
          if (record->event.pressed) {
            add_oneshot_mods(MOD_BIT_LSHIFT);
          }
        } else {                       // Tap + hold.
          if (record->event.pressed) {
            del_oneshot_mods(MOD_BIT_LSHIFT);
            register_mods(MOD_BIT_LSHIFT);
          } else {
            unregister_mods(MOD_BIT_LSHIFT);
          }
        }
        return false;  // Skip default handling.
      }
      return true;  // Continue default handling.

    // Other macros...
  }

  return true;
}
```

leg7, 2025-04-25:
@getreuer Ok, thank you very much for your time and help 👍