vial-kb/vial-qmk #864: vial uses the version of the driver that is compatible with via

jeromeli002, 2025-03-09:
Does vial still use the V2 version of via driver? The web version of via is really not as good as the web version of via. The firmware of vial is recognized as V2 version in via. If it is the V3 driver, how can it be recognized as V3 version? If it is V2 version, when can it be updated to V3 version?

xyzz, 2025-03-09:
it's diverged to the point it's not really compatible with VIA neither is it feasible to change to make it compatible; in a future version we will remove support for VIA (sideloading, etc) altogether as nowadays it's just contributing to the support issues on discord and does not really benefit the end user experience

closed 2025-03-09 by xyzz

jeromeli002, 2025-03-09:
Is there a plan to optimize the web version of vial? It would be best if it supports international languages. The current web version is much worse than the desktop version.

xyzz, 2025-03-09:
it already supports layouts: 

[image]

not sure what you mean by "international languages", I don't think there's a difference to the desktop version here

jeromeli002, 2025-03-09:
[image]

[image]In this way, the interface language is listed separately. The interface language can support multiple languages. In addition, the newly added user key of the web version cannot display the name set by yourself like the front version, but still displays userXX.

xyzz, 2025-03-09:
vial desktop doesn't support localization either, it would need to be implemented first

jeromeli002, 2025-03-09:
Is there a support plan? If I simply change the language in the code, the desktop version is fine, but the compiled web version still doesn't work and will turn into a box. If the keyboard name is in another language, it will also be displayed as a square box. Is it because there is no font for the language? Is there a way to compile it in?

xyzz, 2025-03-09:
you'd have to replace the font in the Qt webassembly build; I believe it's in
```
wasmfonts.files = \
    ../../../3rdparty/wasm/Vera.ttf \
    ../../../3rdparty/wasm/DejaVuSans.ttf \
    ../../../3rdparty/wasm/DejaVuSansMono.ttf
wasmfonts.prefix = /fonts
wasmfonts.base = ../../../3rdparty/wasm
RESOURCES += wasmfonts
```

but i haven't tried it

jeromeli002, 2025-03-09:
Which file is this added to? I'll try it out when I have time.