vial-kb/vial-qmk #878: How to implement advanced Combos

tamura55, 2025-04-14:
I want to implement [key repress Combos](https://docs.qmk.fm/features/combo#customizable-key-repress) like CMB_ALTTAB.

But Vial GUI does not support these advanced Combo settings.
Also, Vial does not support Combos in keymap definition that previously discussed on [Issues #766](https://github.com/vial-kb/vial-qmk/issues/766).

So it seems that advanced Combos cannot used on Vial. Is there any way to implement them?

tamura55, 2025-04-15:
I found [the commit](https://github.com/vial-kb/vial-qmk/commit/0fd99096577a57453f070f2dd94baefa09e369a4) that add combo key repress feature.

In this commit, many changes are made to files in "tests" folder.
I don't know how to use these files because they are not referred when compiling.
Also, I'm not sure if it is in a usable state as it is in "tests" folder.

Does anyone know anything about this?

ingvardm, 2025-05-13:
The files in test relate to testing, you don't need those. So you only need to patch the proccss_combo.c file. Also you can just fork this repo and pull that commit to your fork.