vial-kb/vial-qmk #888: Not able to configure advanced tap dance on Vial

mhalano, 2025-05-15:
I have a configuration on QMK that I couldn't configure on Vial.
I have an advanced tap dance that when I press CAPS once, it enables CAPS WORD. But if I press the same key twice, it enables the regular CAPS. After that, I can press once more and it will disable.
I would like to do the same with Vial, but it seems Tap Dance must be configured over GUI.

Here is my repository: https://github.com/mhalano/vial-qmk/tree/vial/keyboards/xbows/nature/keymaps. My QMK code is under `mhalano-qmk` and it compiles. The code I'm changing is on `mhalano-vial` and it doesn't compile yet with the error: 
```
➜ qmk compile -kb xbows/nature -km mhalano-vial
Ψ Compiling keymap with gmake -r -R -f builddefs/build_keyboard.mk -s KEYBOARD=xbows/nature KEYMAP=mhalano-vial KEYBOARD_FILESAFE=xbows_nature TARGET=xbows_nature_mhalano-vial VERBOSE=false COLOR=true SILENT=false QMK_BIN="qmk"

avr-gcc (GCC) 14.2.0
Copyright (C) 2024 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text    data     bss     dec     hex filename
      0   25638       0   25638    6426 xbows_nature_mhalano-vial.hex

Compiling: quantum/keymap_introspection.c                                                          In file included from quantum/keymap_introspection.c:9:
./keyboards/xbows/nature/keymaps/mhalano-vial/keymap.c:94:6: error: ‘TD_CAPS’ undeclared here (not in a function); did you mean ‘CL_CAPS’?
   94 |     [TD_CAPS] = ACTION_TAP_DANCE_FN(on_tap)
      |      ^~~~~~~
      |      CL_CAPS
./keyboards/xbows/nature/keymaps/mhalano-vial/keymap.c:94:6: error: array index in initializer not of integer type
./keyboards/xbows/nature/keymaps/mhalano-vial/keymap.c:94:6: note: (near initialization for ‘tap_dance_actions’)
 [ERRORS]
 |
 |
 |
gmake: *** [builddefs/common_rules.mk:365: .build/obj_xbows_nature_mhalano-vial/quantum/keymap_introspection.o] Error 1
```
I don't know for sure how implement this on Vial, since it mentions QMK documentation, and I did my changes for QMK.

mhalano, 2025-05-15:
This would be done in firmware itself or in the OS? Do you have any link for the documentation about how to set a keycode? As you can see by my repository, my knowledge is based on QMK, so I'm not aware of differences from QMK and Vial.

mazharenko, 2025-07-21:
Please, introduce a switch to disable VIAL tapdance. There are already  many keyboards inherited from QMK that rely on defining tap_dance_actions